polyend/TrackerBetaTesting #12: [1.1.14] Tracker MIDI over USB out of sync

Jacknametrouble, 2020-09-09:
Hi. I'm currently working on a project using the tracker connected to Garageband via USB on an iPad Pro with a SATECHI USB-C adapter.  I've set all the MIDI-settings on the tracker/config/MIDI to USB where applicable. I immediately noticed after 2-3 bars of playing the Garageband-project on my iPad, the tempo eventually got misaligned. To be sure this wasn't just an issue with the iPad, I tried flashing the tracker back to v1.1.0. and the BPM-problem does not occur

Additional questions to answer:
Does this bug happen every time or just once? - as described,
Does this bug happen in previous firmware versions? -  Not present on 1.1.0
Does this bug happen in every project? -  not relevant,

Jacknametrouble, 2020-09-23:
[1.1.16] -  confirmed. Ableton Live MIDI over USB to Tracker (sync + transport) > send clock > activate live rec when the metronome is on > after some time 60 - 80 sec the metronome clicks drifts away from each other in a noticeable manner.

shuler323, 2020-09-23:
similiar thing in #7, please check upcoming version (1.1.17) when it'll be available

closed 2020-09-23 by shuler323