polyend/TrackerBetaTesting #17: [1.1.14] inconsistencies in the rendered audio file.

Jacknametrouble, 2020-09-09:
I had inconsistencies on the exported WAV when rendering a song internally; somehow the pattern order was scrambled and also some sounds were muted. When playing back the song live, it is properly played.

Jacknametrouble, 2020-09-10:
Another similar comment from another user:

Miltos-S, 2020-09-19:
I can confirm I also experience this issue. The exported file has a slightly faster tempo than the playback/realtime recorded audio.
(They don't align together if you try to sync them in a DAW). Also when I stem export and then import the files/stems to Cubase, they appear to go out of sync so there is some tempo inconsistency between them too. (there is audible flamming and the stems don't play well together)  The workaround, for now, is to realtime record each stem manually by soloing/muting, etc. I'll try to provide some example files for this issue.

Miltos-S, 2020-09-25:
I have prepared two examples that probably fit under this issue.

The first one is about the difference in the tempo of the exported file in relation to the playback tempo.
In this example, I have a demo song, set at 120bpm in the Tracker, that is realtime recorded in the DAW(Cubase) and also exported from the Tracker.
The recorded one roughly equals 119.94 bpm in Cubase
and the exported one roughly equals 120.13 bpm in Cubase.
So if we play the two files together end up being out of sync and with different duration.
[Different tempo between Export and Playback.zip](https://github.com/polyend/TrackerIssues/files/5279916/Different.tempo.between.Export.and.Playback.zip)

Miltos-S, 2020-09-25:
The second example is about an inconsistency in tempo between stems.
It seems that when exporting stems the tempo slightly fluctuates randomly for each stem, so if we then put all the stems to play together in a DAW we hear some "flamming". They are slightly out of sync so we lose the "punch". In a complicated track, this inconsistency may cause the song to lose its groove.
In this example, I made a simple demo of four clicky sounds playing together in 4/4, like a metronome.
One file is the mix export which sounds as it should with all the sounds playing together in good sync.
The other file is the four strem exports summed later in Cubase. In this file, we can hear that the sounds play slightly out of sync as I describe above. I have also included the Tracker project file.
(I should also note that I have tried to manually record stems by soloing/muting tracks and in that case, I didn't notice this issue.)
[Stems summed in DAW.zip](https://github.com/polyend/TrackerIssues/files/5279969/Stems.summed.in.DAW.zip)

piotr3500, 2020-10-01:
Audio vs. Render might differ in time just a bit. Please use one or another.

closed 2020-10-01 by piotr3500