polyend/TrackerBetaTesting #18: When two Trackers are MIDI chained > ability to change their patterns simultaneously.

Jacknametrouble, 2020-09-09:
When two (or more:) Trackers are MIDI chained > ability to change their patterns simultaneously, maybe by adding a Program Change function in the MIDI update?

Explodicide, 2020-10-23:
Yes, I would love this.  I'm using the Tracker with Elektron grooveboxes, and being able to have one drive the patterns of the other would be fantastic!

If you had the option of setting Tracker to respond to a Pattern Change command by switching to a new pattern and starting it from the beginning, it would let you overcome the limitation of only one pattern length by having another device telling the Tracker when to change patterns.

tc-joegiralt, 2021-05-05:
this would be great.

closed 2021-12-02 by Jacknametrouble