polyend/TrackerBetaTesting #19: [1.1.14] Tracks manually recorded to DAW (Tracker as a MIDI slave) drifts from each other.

Jacknametrouble, 2020-09-10:
While recording a new track to disk, track after track.

Here is the setup: Use Live 10 for master clock/ transport > the Tracker via USB, I solo one track, record it, then go on to the next track on another channel etc.

The problem is the Tracker since to drive in time so none of the Track is correctly synced, and it seems that it keeps drifting as after I recorded four of the channels I now have to edit every bar so they sync up.

It was working way better on the last firmware. I will try exporting the stems to check if the problem is present here as well.

Additional questions to answer:
Does this bug happen every time or just once? - Always,
Does this bug happen in previous firmware versions? - It was working way better on the last firmware,
Does this bug happen in every project? - On the ones I've tried,

shuler323, 2020-10-01:
@Jacknametrouble there was a fix with ext clock in 1.1.17, regarding to #7 and #12, is bug still present in 1.1.17?

Jacknametrouble, 2020-10-01:
@shuler323 no user reports in regard to this issue/fix.

Jacknametrouble, 2020-10-01:
@Stazma any changes? (that was your report as far as I remember),

Stazma, 2020-10-01:
> 
> 
> @Stazma any changes? (that was your report as far as I remember),

I did not experienced any trouble with this but I havn't really tested either. I'll try to run a real test about it today if I have some free time for it and report back asap.

Stazma, 2020-10-02:
Back on it today, I'm having a super hard time syncing the Tracker the Live, it just keep freezing almost every minutes. Maybe it's the project I'm working on so i'll try with a fresh one and report back.

Stazma, 2020-10-02:
Ok so I just ran a quick test with a three different channels of HiHats in a new project, still firmware 1.1.7, synced via USB, everything seems tight. I can see some minor timing offset if I zzom like crazy but nothing to be heard really.

[image]

shuler323, 2020-10-05:
cool, so I'm closing this issue

closed 2020-10-05 by shuler323