polyend/TrackerBetaTesting #31: Chord / Arp FX command

MegaTurrican, 2020-09-11:
I am sure you guys have already thought about that, the Chord / Arp FX command inspired from LSDJ, would make a great addition for us oldschool chiptuners.

[image]

jemmons, 2020-11-07:
+1 I'd love some sort of arp functionality. Right now I fake it by really cranking up the BPM (which makes the rest of pattern a pain to work with) or rendering my arp sequence to a new instrument (which locks it to a tempo). An FX command would be ideal though this could also be achievable with finetune automation if finetune had a greater range and LFOs were more customizable? Or a new Instrument Parameter just for arp? Or even a roll if we could somehow give intervals to the roll in addition to step count?

Madrayken, 2020-11-30:
Totally agree. I can't think of a tracker I've used which doesn't have some kind of note-arp. It's especially important in trackers, as each note chokes the previous one, meaning you run out of 'real' voices fairly quickly.

RadicalAns, 2020-12-01:
+1 to this idea. I'd love to have this type of functionality.

tilllt, 2020-12-27:
I dont know LSDJ but it sounds as if it is similar to what i meant in my suggestion.

fragment23, 2020-12-29:
+1 for the arp fx

MetabaronCZ, 2020-12-31:
I would like to see that too. There is Roll Fx which makes substeps, so it could be possible to extend this functionality.

eholzbach, 2021-01-24:
Was thinking today I'd like to arpeggiate midi gear.

Jacknametrouble, 2021-12-02:
implemented

closed 2021-12-02 by Jacknametrouble