polyend/TrackerBetaTesting #35: 1.1.14 Mute states after solo/unsolo

stevesomeone, 2020-09-11:
If you have a mute group set up, then solo a track, the Tracker defaults to all tracks active after unsolo. It seems to reset all applied mutes to active.

Does this bug happen every time or just once? All the time
Does this bug happen in previous firmware versions? Yes
Does this bug happen in every project? Yes

([https://www.dropbox.com/s/ve0yq80djzd8d97/Solo%20and%20mute%20groups.mp4?dl=0])

shuler323, 2020-09-16:
Thank you for suggestion, but i'd rather set 'enhancement' label for this. We will discuss possibility of implementation

birdydy, 2021-01-18:
Old topic already, but I second the idea ! I think it's more convenient if the previously muted tracks don't all reset to "active" after you've soloed and unsoloed a track. It is basically deterring you of using solo once you've set some mutes.

closed 2021-12-02 by Jacknametrouble