polyend/TrackerBetaTesting #43: Hex display of pattern position

ambv, 2020-09-16:
## Decimal positions
Currently pattern positions are decimal and indexing starts at 1:
[image]

## Hexadecimal positions
I'd like to propose a traditional hexadecimal position numbering with indexes starting at 0:
[image]

## Advantages
The most important advantage of this design is that when using 4/4 meter, all beats are on power-of-two values. In the screenshot you see beats of the first bar on 00, 04, 08, 0C. What would be the second bar? 10, 14, 18, 1C. Third bar? 20, 24, 28, 2C. Fourth bar? 30, 34, 38, 3C. And so on. As you can see, there's a pattern. Bars start at 00, 10, 20, 30, and so on. Divisions inside bars have predictable values.

Compare this with the current situation: first bar beats are 1, 5, 9, 13. Can you guess what the second bar beats will be? 17, 21, 25, 29. Third is 33, 37, 41, 45. Fourth is 49, 53, 57, 61. Hard to predict, no discernible pattern.

The hexadecimal notation is natural for existing users of tracker software.

## Implementation
* The feature would be toggleable in Config.
* The feature is global (per device, not per project).
* This feature request only discusses hexadecimal pattern position.
* For consistency it would make sense to also make pattern numbers themselves hexadecimal, too.
* All other numeric values in the Tracker software, in particular instrument indexes, FX values, and automation, would stay decimal, displayed as is today.
* The change would require display of pattern positions and pattern numbers to be consistently hexadecimal on all screens, including button labels on the bottom part of the screen.
* Leading zeroes are displayed to help users tell that the hexadecimal mode is enabled.

closed 2020-09-24 by ambv

pawal, 2020-10-21:
Was this issue closed because it has been implemented, or because it was deemed not interesting?

ambv, 2020-10-22:
It's fixed.