polyend/TrackerBetaTesting #44: [1.1.14] Position FX command doesn't work with loops and reverse playback

ambv, 2020-09-16:
See attached video.

1. First it shows the entire sample in Pattern 1.
2. Then it shows proper position selection with the `p` command in Pattern 2.
3. Then still in Pattern 2 it shows the `p` command not working when the sample playback is set to loop forward instead of single shot. That's **problem 1**.
4. Then in Pattern 3 it shows the `p` command behaves erratically when `r<<<` is used:
  - it does not change the starting position of reverse playback;
  - but it *does* affect the duration of playback (note the sample being cut sooner when `p` greater than zero is selected).
That's **problem 2** which is a specific case of FX1 and FX2 not working in tandem.

### Attachments and links

https://www.dropbox.com/s/a0k49vwnaz3712f/PositionFX-reverse.mov?dl=0

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
every time

* Does this bug happen in previous firmware versions?
yes, tested since at least 1.1.5

* Does this bug happen in every project?
yes

piotr3500, 2020-10-01:
It does work, but you cannot push further if loop end hits sample end.
You can check this in Sample Playback (loop mode).

Reverse FX changes start point to an end point and goes backwards.
Wheres, the Position FX changes the start point, not end point.

ambv, 2020-10-01:
To be addressed after 1.2.