polyend/TrackerBetaTesting #47: [1.1.14] Consistent crackling present in stereo exports.

Jacknametrouble, 2020-09-17:
There is a subtle but consistent crackling present in stereo exports that were not present in the v1.1.0 firmware.

It is happening in the change from pattern 11 to pattern 12 when playing the track on the tracker in "song" mode too.

Project + workspace + reference mix
https://drive.google.com/file/d/1eWon0KE-C96qqOgdq0vOoh5xquGP9td7/view?usp=sharing

* How often does this bug happen? (every time, often, "it only happened once") - always,

* Does this bug happen in previous firmware versions? -  wasn't happening on the v1.1.0 (I jumped straight  from 1.1.0 to 1.1.14),

* Does this bug happen in every project? - on the several ones I've tried,

Unpaid-Intern, 2020-09-17:
Hello! I was the one to originally report this bug to Jacek. Please let me know if you need me to submit additional information/patterns to try to reproduce this error. Thanks!

Miltos-S, 2020-09-19:
This issue happens here also. There are some consistent crackling/clicks on exports (stems or stereo). They are more evident if I apply some EQ to the stems (in the DAW) and boost high frequencies. The crackling might appear all over the file but there are also clicks at the start of music-samples/notes or between them. On realtime playback, this crackling and clicks is not happening so for now the workaround is to realtime record the stems or song. I'll provide example files asap.

Miltos-S, 2020-09-19:
There are also clicks in the audio preview during an export but these don't appear in the rendered file and are a different thing from what I describe above.

MegaTurrican, 2020-09-24:
Since issue #62 was duplicated to this one, please note that the clackling does not only occur in exports, but also when playing back a song. It's present even more when there are many wavetable synths in a pattern and also the problem remains with sample clicks & crackles when programming a busy pattern with many effects, rolls etc. I think this should be of high priority to fix, because getting bad exports and random clicks in the audio output is an absolute dealbreaker for an instrument.

MegaTurrican, 2020-09-24:
Here is my original submission and Tracker project file for reference:

I've been trying to export a track clean but unfortunately I get lots of clicks / sample start envelope crackles, on the final wav.
The clicks are also heard during playback actually. There is a lot going on in the track, so I attached it for reference.

Same clicking issues have been occurring in all firmware versions in this track, with only exception the 1.1.0 version (which we used this track for promo). It seems that busy tracks / dense sample alternations might cause this.
[qebo-002.zip](https://github.com/polyend/TrackerIssues/files/5275496/qebo-002.zip)

Miltos-S, 2020-09-24:
> Here is my original submission and Tracker project file for reference:
>
> I've been trying to export a track clean but unfortunately I get lots of clicks / sample start envelope crackles, on the final [...]

I have downloaded this track to check it out.
It does have clicks in both playback and export.
About the playback, I have soloed each track to investigate more. Clicks are mostly audible in the second track that contains the kick sounds so clicks are more evident there due to the low frequencies. I added some note-offs in the first pattern to see if it fixes it and it seems that it does eliminate most clicks. Though you may still hear some random clicks. If there is a way from the firmware to have a small fade-out by default when a sound from a step is interrupted from the next step, it might help in this kind of situation but this is something for the developers to see if it's possible.     
I have attached a recorded playback and an exported track with firmware 1.1.16 for you to compare with your unit.  (Unfortunately in mp3 due to file size).
[qebo Audio test.zip](https://github.com/polyend/TrackerIssues/files/5278703/qebo.Audio.test.zip)

Miltos-S, 2020-09-25:
I have prepared an example of the crackling type of sound/noise in the exported files.
When recording the sounds in realtime this noise does not exist.
For the effect to be more evident all the sounds (both recorded and exported) have a mid-high frequency boost with EQ.
The issue is very audible in the synth example.

The clicks that also exist in the drums example are a different thing and most likely they could be fixed if I have added some note-offs or tweaked the envelopes a bit more.
The problem is that in some cases its not possible to eliminate these clicks (eg there are no empty step to add a note-ff) so
it would be nice if the Tracker added small fade-outs by default when a sound is interrupted by a next step causing a click. 
[Example of Crackling in Exports.zip](https://github.com/polyend/TrackerIssues/files/5279825/Example.of.Crackling.in.Exports.zip)

thebrackett, 2020-09-25:
Can also confirm that I have narrowed some of my issues down to the use of roll type of "Rv" I used Rv8 near the end of a pattern and ran into the issue.

thebrackett, 2020-09-29:
Here's another example of the issue. You can hear it in the very first pattern (aka top of the reference mp3).

* The issue is crackling/zippering effect. 
* NOT present in normal playback, but definitely present in Stereo export. 
* Occurring in instrument 1 which is a wavetable synth instance.
* Occurring in a very simple pattern with only mono line played on single instrument

[PT_bug_report_export_crackles.zip](https://github.com/polyend/TrackerIssues/files/5296460/PT_bug_report_export_crackles.zip)

closed 2020-09-30 by andrzejmagierski2

thebrackett, 2020-09-30:
Woo hoo! Nice going y'all! ♫♪.ılılıll|̲̅̅●̲̅̅|̲̅̅=̲̅̅|̲̅̅●̲̅̅|llılılı.♫♪