polyend/TrackerBetaTesting #50: [v1.1.15] Sample Preview

djr005, 2020-09-19:
## Hi, sorry you had trouble!

### Steps to reproduce

Hi Guys, I loved in previous versions, we could continually hold down the "preview" button and cursor down using the direction arrows to preview each sample when auditioning samples to load. However, in the latest version, we can't seem to be able to do this. Cheers 

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
Every time
* Does this bug happen in previous firmware versions?
No
* Does this bug happen in every project?
Yes

Jacknametrouble, 2020-09-19:
We're aware, this functionality will soon be back. Thank you.

djr005, 2020-09-19:
Thanks Jacek 👍

closed 2020-09-21 by ziej8asek