polyend/TrackerBetaTesting #51: PageUp / PageDown For Key Scrolling

rpc9943, 2020-09-19:
In older software such as impulse tracker I could navigate much faster especially for longer patterns using a pgup/pgdn type key command that could skip. The skip function is great but currently only works when entering a note. Unless I am mistaken with the current interface we don't have a means to navigate quickly when in note entry mode for example (we cant turn wheel to move quickly, it changes note information). If we could have a command hotkey to navigate quickly up or down a page would  be a welcome addition to the current navigation.

djr005, 2020-09-19:
Totally agree. We have Top already, but a Top/Half/Bottom key commands would be nice.

BerndPeuser, 2020-11-20:
My idea is to that enhancement is, if you edit the Keystep, lets say 16 and then press Cursor key down it move the cursor 16 steps down and not only one.

danapeks, 2020-11-22:
This would be really helpful. In Renoise you can do it by in increments of 25% of the pattern length

tfspsound, 2020-11-24:
I originally thought that this was how Step worked, on use I realised it only applied to recording notes. Like Bernd said ^ Let Step control the cursor movement or set Shift + Down / Up to move the cursor 16 cells. Instead of 16 make it user definable in Config? I also think being use Shift + Jog Wheel should increase / decrease values quicker

closed 2021-12-02 by Jacknametrouble