polyend/TrackerBetaTesting #55: Additional randomization capabilities using the Fill tool

modmodmodmod, 2020-09-19:
I hope the Fill tool can randomize ADR of instrument volume, filter respectively and start and end of sample playback. 

And I would appreciate it if I could set the pitch envelope of instruments, so I would like to request these if possible.

piotr3500, 2020-10-21:
You can control the filter and start of the sample by using dedicated FXs.
We are thinking about adding ADR and sample end too.

closed 2020-10-21 by piotr3500

modmodmodmod, 2020-10-21:
Thank you.  I use filters and sample start a lot.  I also loop play back the samples as decay.  Still, I think it would be great if you could add volume envelope to the fill feature.  As you may already know, decay and release are especially important to composition of the songs :)