polyend/TrackerBetaTesting #57: Live monitoring line in

theGh0stCat, 2020-09-19:
## Hi, we're excited to hear your idea!

Before you start, did you check whether your feature was already requested?
Searching might save you quite a bit of time here.

### A new feature

Allow option for audio from line in to be heard at all times.
I’m running a mobile setup with tracker midi controlling a microfreak who’s audio is going in to the line in of the tracker. A good power brick allows me to take these two units wherever I so please and jam to my heart’s content... however, with this setup, if I’m in pattern mode on tracker sequencing midi notes for microfreak to play I can not hear the audio from the microfreak. To hear your sequenced results then you have to go in to sample recorder mode(great new feature btw). An option to monitor line in would allow easier midi sequencing when running a closed loop of gear, this would also open a few doors for live performances centered around the tracker.

HakkahFR, 2020-10-03:
The most important feature for me

jaap3, 2020-12-27:
I believe this is now possible with firmware version 1.3.0

Jacknametrouble, 2021-12-18:
This is possible since possible with firmware version 1.3.0

closed 2021-12-18 by Jacknametrouble