polyend/TrackerBetaTesting #60: [1.1.15] It may not be possible to add slots in song mode.

modmodmodmod, 2020-09-20:
The frequency was about once an hour.
 So far it has been confirmed in only one project.

In the video linked below, it disappeared as I went back and forth between other screens, but the pattern mode display is also strange.  There were times when it froze in this state and the power could not be turned off.  At that time, I unplugged the power cable and restarted, and it worked.

[https://drive.google.com/file/d/1r6W4ZLnM_rWrjMdWu_ZaXIX0pSffmKR0/view?usp=drivesdk](https://drive.google.com/file/d/1r6W4ZLnM_rWrjMdWu_ZaXIX0pSffmKR0/view?usp=drivesdk)

pigeonsky, 2020-09-20:
Reverted back to v1.1.0 and its working in this version. THEN, went to v.1.1.12 and now the PT is stuck on the load screen (just showing the logo...)
Help?

piotr3500, 2020-09-21:
Sometimes screen buttons (3 and 4) freeze. It's not only related with Song screen, it's in every screen. 
We're on it.

piotr3500, 2020-09-21:
Duplicate of #67

closed 2020-09-21 by piotr3500