polyend/TrackerBetaTesting #65: [1.1.15] Pressing the down cursor button in the cofig menu during pattern playback, PT freezes.

modmodmodmod, 2020-09-21:
Infrequent.  It was about twice a day.
 So far only one project has been confirmed.

In the video linked below, I was only able to shoot it was frozen, but this same situation occurs frequently.  The power button doesn't work all the time, so I always unplug the cable to release it.

[https://drive.google.com/file/d/1UudSvOxF4vcJ3QY4RDNQuUrOdvjliCVH/view?usp=drivesdk](https://drive.google.com/file/d/1UudSvOxF4vcJ3QY4RDNQuUrOdvjliCVH/view?usp=drivesdk)

Jacknametrouble, 2020-09-21:
Cannot reproduce the issue on the General/Root note nor any other Config positions. Does this always happen in this particular project? Please try deleting the hidden "workspace" folder located in the SD card root note. Check if it's still happening, if so - please zip andlink this particular project so we can take a look at it. Thank you.

modmodmodmod, 2020-09-21:
Thank you for reply.  I have tried to reproduce it but now this freeze does not occur.  I'm using MacOS 10.15.3 to access my SD card, how can I view the invisible workspace folder?

closed 2020-10-01 by piotr3500