polyend/TrackerBetaTesting #66: [1.1.15] sampling hiss or white noise.

theGh0stCat, 2020-09-21:
## Hi, sorry you had trouble!

Recording external sources has static in 1.1.15
Running off power bank and sampling off iPhone. Lower quality from 1.1.14 with the same gear.
### Steps to reproduce
Record external gear
### Attachments and links

Often the best way to show the problem is to record a short video,
attach the Tracker project you were working on, and provide links.

Pro-tip: You can drag and drop images and `.zip` files directly to the edit
box when writing your bug reports. Videos are too big as attachments, you
will have to use an external service for this and provide us links. Please
don't use services that delete files after short periods of time.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")

* Does this bug happen in previous firmware versions?

* Does this bug happen in every project?

piotr3500, 2020-09-21:
Perhaps it's a cable problem. 
We can't reproduce it.

Also, please delete unnecessary text when posting an issue. Thanks!

theGh0stCat, 2020-09-21:
Thanks for the reply, sorry for the unnecessary work.
I had the master volume up a bit loud. I have noticed a strange
relationship between master volume and the line level for recording. My
iPhone can have volume all the way up without clipping as long as tracker
master volume is around 50. If tracker master volume is at 70 than iPhone
clips at full volume. This is very different from the volume control levels
and relationship between overall volume and sampling volume that I have
seen in the past. Is this the intended relationship?

theGh0stCat, 2020-09-22:
Sorry to bother you, but I am unable to update to the latest 1.1.16
firmware. I swapped the firmware on micro sd card... but the tracker
freezes 10/10 times when in config mode and attempt to go lower than the
midi page. Requires pulling power plug to reset. Any ideas?

Jacknametrouble, 2020-09-22:
@theGh0stCat 

The press of the reset button is flashing the Tracker with a random firmware version chosen from the ones available in the Firmware folder on the SD card.

There's also another way of emergency flashing in case anything would go wrong:

- remove the SD card from the Tracker SD card slot,
- power up the Tracker,
- run the Polyend Tool v1.1 (polyend.com/downloads) and detect the Tracker,
- choose the firmware iteration you wan to flash the unit with,
- press the hidden reset button located on the back panel of the instrument,
- wait till the progress bar in the app will reach 100%,
- if the Tracker won't reset automatically, reset it manually,
- place the SD card back in the Tracker's SD card port,

closed 2020-09-22 by Jacknametrouble