polyend/TrackerBetaTesting #70: Loop Crossfading for samples

Intothegroove, 2020-09-21:
I would love to have some kind of simple loop cross fading for samples, currently most of the time you still get clicks when trying to loop a sample, even when the waveform is at zero crossings. thanks!

danapeks, 2020-10-19:
Yes this is essential

ROM72, 2020-11-01:
I agree,
it would be nice to have a crossfade setting at the start and end of the loop.
For example a crossfade option that you set a bit like ableton sampler loop mode, with one knob you control the crossfade duration of both start and end, it's very convenient.

drowsy-net, 2021-01-23:
Was going to add this as a feature request, but here it is. Forward/ping-pong are really only useful for perfect loops - but wavetable playback already makes use of those. This single feature is what changes Tracker from a one-shot player into something capable of actual sample synthesis. Hope this gets picked up!!

mxactvtd, 2021-01-28:
I would say ping-pong mode is even worse for perfect loops, even with just a sine wave with perfect sample accurate 0 in beginning and end you get clicks in all modes, single-cycle solves the clicks but has it´s own problems, probably linked to the length value having too little precision, will add it as its own issue.. thanks for amazing hardware and this github issues page is a perfect way to improve it!

JonathonMSmith, 2021-09-09:
Could this be extended to the granular mode?