polyend/TrackerBetaTesting #72: Audio stops working randomly

anemlife, 2020-09-21:
Already happened multiple times. Audio stops workings, everything else works, I can navigate in the menus etc... I save my project, switch the tracker off and on then it works again. 

Noticed this can happen when previewing samples from the sd too fast but can't reproduce the other cases where it happened. (can't do a video today, maybe during the week)

Jacknametrouble, 2020-09-21:
Does this happen on any particular project? If so, can you please link it here?

anemlife, 2020-09-24:
Issue was due to a broken .wav file that caused the audio to crash when previewed. Can't reproduce the other case where it happened but it looks to work with 1.1.16 so I close this issue.

closed 2020-09-24 by anemlife