polyend/TrackerBetaTesting #74: Option for glide to bend active note rather than triggering new note

djkurzman, 2020-09-22:
Tracker currently re-triggers the instrument sample when activating glide FX. Requesting option prevent retrigger of the sample, and rather portamento the actively playing sample to the new note when glide is used. This would make the glide effect similar to a 303 slide.

HakkahFR, 2020-10-03:
Exellent suggestion and very important for me too

piotr3500, 2020-10-21:
Slide FX is coming.

HakkahFR, 2020-11-19:
> Slide FX is coming.

Piotr, are you still working on the fx slide or it was canceled ? Ty for the last update with the live line input, change everything for me

piotr3500, 2020-11-19:
Sure, more FXs are coming, not very soon tho.
Working on some other stuff ATM.
I'm quite sure you'll like it too ;)

Tombelliard, 2021-01-21:
Does slide fx will be available for midi too?

djkurzman, 2021-01-21:
It would be interesting to see if that could work with midi as well.
Generally, if the external MIDI device supports it, you would have to have
an overlap of notes to slide from the first to the next. Currently this can
be done in the Tracker by using a couple of tracks and doing some skillful
note placements and cuts.
Portamento slide with a sampler means placing the slide effect on the note
you want to slide to. Doing the same for MIDI notes could just be a matter
of making sure the Tracker doesn't end the previous note until the next
note without slide is played. Although, I'm not an expert on MIDI, so it
could be easier said than done.

djkurzman, 2021-04-27:
Playing with the new Slide FX in 1.4.0 Beta 1. Hopefully this was not the feature to address this request. The new Slide feature is a constant note bend on a newly triggered sample. Whereas this request was to be able to trigger a portamento on an actively playing sample. Don't get me wrong, I do like the new slide because it is very useful. I just want to make sure the portamento request remains open. Thanks!

HakkahFR, 2021-05-11:
That's the biggest problem of every synth i use, the portamento is not musical. Tb 303 and sh 101 sound so good because the note are not retriggered. I really don't understand why devellopers don't make the possibity to switch betwin the two mode of slide : retrigger and non retrigger. I hope for this change in the tracker, it's one of the most forgotten option of all hardware

HakkahFR, 2021-05-25:
Its finally ok i made a mistake by entering the slide fx on the note instead of between two note

djkurzman, 2021-05-26:
Same here, and I admit that I didn't read the description panel well enough to fully understand how SlideFX was working. It still seems maybe a little unintuitive for use as a portamento. I do like the slide effect as it is, but maybe just for creating continuous up and down slides for samples and being able to set the value to control transition speed rather than 16th semitones. A normal portamento would be better as [Set first note playing, Set second note with portamento FX (value effects transition speed), portamento FX slides previous note to new note]. This would be exactly how GlideFX currently works, except without retriggering the sample.

It would also be cool if that effect could trigger portamento over MIDI, simply by allowing the previous note to cut slightly after the next note begins playing. That way we do portamento slides with external 303 synths and such without having to use up two channels.

HakkahFR, 2021-05-26:
100% agree with you djkurzman

djkurzman, 2022-04-19:
I hate to sound pushy, but is there any word on where a proper portamento is on the roadmap? Unfortunately, I have barely used the Tracker while waiting for it. The currently glide and slide fx are just not as functional in my opinion.