polyend/TrackerBetaTesting #76: [1.1.16] Loud Popping Through Speakers/Headphones Upon Power On/Off

AirDnB, 2020-09-22:
### Steps to reproduce

Turning the PT on and off produces a painfully loud sound through whichever speakers or headphones are connected.  

https://photos.app.goo.gl/8D8XjhrL4dECWewT7
**_Please click the comments section ar the bottom for comment on each video_**

This has occurred since I first switched on the unit with the firmware it shipped with (1.1.10?)

Happens in all firmware versions

Happens any time the PT is powered on or off.

Jacknametrouble, 2020-09-22:
This is a hardware design thing and it's impossible to fix this with a firmware change, unfortunately. This is only present in the first hardware batch and is not happening in the next ones.

closed 2020-09-22 by Jacknametrouble

AirDnB, 2020-09-22:
Thats...great for the new owners.

Jacknametrouble, 2020-09-22:
Hey, apologies for the inconvenience, we're still learning. Building musical instruments from scratch is not a piece of cake as you may expect. Sometimes you need to learn on your own flaws. Hope it's not like a fun killer for you.