polyend/TrackerBetaTesting #80: 1.1.16 Loss of volume

muhfugga, 2020-09-22:
## Hi, sorry you had trouble!

### Steps to reproduce

When in sample loader, I was previewing my recordings. I already had 5 samples loaded and sequenced in patterns 1,2, and 3. When in sample loader, I was previewing my recordings. I have about 70 recordings. After previewing around 8 of them the recordings would not preview but also I could not get any sound from the tracker at all. I eventually powered off my tracker and powered back on and I then could hear my pattern. I went back to preview my recordings and the same happened where after previewing and scrolling through about 7-10 recordings I would lose sound and not be able to gain it back unless I powered off and back on. This happened a few times in a row until I figured out which recordings I wanted to load and didn't have to preview as many. 

### Attachments and links

Often the best way to show the problem is to record a short video,
attach the Tracker project you were working on, and provide links.

Pro-tip: You can drag and drop images and `.zip` files directly to the edit
box when writing your bug reports. Videos are too big as attachments, you
will have to use an external service for this and provide us links. Please
don't use services that delete files after short periods of time.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")

* Does this bug happen in previous firmware versions? I usually don't reuse my recordings so this is the first time I have tried this so I'm not sure if it's this particular firmware that caused it.

* Does this bug happen in every project?

ziej8asek, 2020-09-22:
Duplicate of #78

closed 2020-09-22 by ziej8asek

piotr3500, 2020-09-22:
@muhfugga please delete all the necessary text when posting an issue next time. 
It's only there to give you hints. 
Thank you!

muhfugga, 2020-09-22:
> @muhfugga please delete all the unnecessary text when posting an issue next time.

LOL...ok i'm sorry. Do you know if the Lo fi sampling is coming on the next 1.2 update?

piotr3500, 2020-09-22:
No, it's coming to 1.3, but shouldn't take as long as 1.1>1.2.

muhfugga, 2020-09-22:
Oh no problem at all. I'm loving this device and will continue to show support. Thank you again.