polyend/TrackerBetaTesting #81: [1.1.16] MIDI Input stuck keys

djkurzman, 2020-09-22:
Having an issue with MIDI Input (have not tested with USB MIDI) where pressing of keys results in a previously held key becoming stuck on. For example, if I play 3 keys simultaneously, all keys produce sound as expected. If I press 3 keys one after the other, giving plenty of time for the previous key is fully released before pressing the other, no issue occurs. When pressing keys one after the other while a previous key is still being released (slight overlap), there is a chance that the new key will only bri
efly produce the new note, while the previous note continues to play even though the previous key is fully released. The old note will continue to play until the new key is released.

The issue is intermittent, but frequent enough to easily reproduce. It first appeared in version 1.1.15. The issue does occur with any project. Testing with multiple MIDI interfaces.

[MIDI Input stuck keys - Polyend Tracker 1_1_16.zip](https://github.com/polyend/TrackerIssues/files/5262582/MIDI.Input.stuck.keys.-.Polyend.Tracker.1_1_16.zip)

Jacknametrouble, 2020-09-23:
Cannot reproduce over MIDI over USB both on the attached project nor any other.

djkurzman, 2020-09-23:
Not sure what's causing it, but it's very easy for me to reproduce on
1.1.16 with any MIDI DIN input. In the video, you can see that when playing
the 3 keys in short sequence, the 3rd note plays briefly, but the second
note continues even though the key was released. It only stops playing when
I release the 3rd key. This doesn't happen with the tracker grid keys, and
only started on version 1.1.15. Prior versions had normal MIDI behavior
with the same setup. Please let me know if there is any extra info I can
provide

djkurzman, 2020-09-23:
I should add that it doesn't happen with all key sequences all the time. You may have to play around to notice which key sequences are having the issue. The example in the video is just one that I happened to find that easily reproduces the issue.

Jacknametrouble, 2020-09-23:
Need to check the DIN scenario, only checked the USB connection. Thank you for the additional commentaries.

Jacknametrouble, 2020-10-01:
Couldn't replicate issue both on the DIN nor USB connections. If it's still happening (and will be happening on the coming v1.1.18) please try to shoot a video so we can take a look at it. Thanks,

closed 2020-10-01 by Jacknametrouble