polyend/TrackerBetaTesting #84: [1.1.16] All buttons have stopped working, other than soft keyboard buttons, jog wheel & power button.

teamoth, 2020-09-23:
In v1.1.16 whilst a sequence was playing I copied and pasted a short sequence of notes. Once the paste completed all buttons other than the keyboard buttons, jog wheel and power button became unresponsive, so I was left with the sequence running with no way to change anything, or stop the sound. 

Turning the Tracker off and back on takes me back to the same project, but the buttons are still unresponsive, other than the keyboard, jog wheel and power button. I can play notes on the currently selected instrument using the keyboard, so audio is still working (although I didn't think to capture that in the video). So it's currently stuck like this. I can turn it on and off, but I can't do anything.

Below is a zip file of the project (um it's pretty simple and a bit odd...):
[aloof end.zip](https://github.com/polyend/TrackerIssues/files/5270294/aloof.end.zip)

Below is a link to a video showing how the buttons don't work (seems this video format is not supported for uploading to github):
[https://photos.app.goo.gl/vQzkqktuHFDFvd9t6](https://photos.app.goo.gl/vQzkqktuHFDFvd9t6)

So, I intend to re-flash the Tracker using the Polyend Tool shortly, as currently I can't do anything with it.

Jacknametrouble, 2020-09-23:
Hey, Thank you for reporting. Just unplug the USB C cable and plug it back in. This will resolve the issue, we're looking for the solution already. Best,

teamoth, 2020-09-23:
Hi Jacek, I didn’t realise that would fix it! Yeah that’s worked great.
Many thanks for your help.
Tim

Jacknametrouble, 2020-09-23:
I'm confirming this bug as I've faced this issue multiple times myself and many other users reported the same thing. The only solution when the controls stop working is to unplug the USB C cable. This is not really replicable, happens in many different use scenarios or "just like that".

closed 2020-09-30 by shuler323

shuler323, 2020-09-30:
i've added some procedures in code to avoid this problem. Please check upcoming version (1.1.18) when it'll be available

williamdiffey, 2021-06-01:
I have just experienced this on 1.4.0beta - removing the USB fixed it. It occured when assigned patterns in song mode but I cannot replicate it.

16bitglitch, 2021-09-25:
Just had this same problem happen while editing samples in 1.4.0. Fixed when I removed USB C.

ssssam, 2022-03-13:
This just happened to me on 1.5.0 while I was recording a note in a pattern. Removing USB C resolved it.

Flipez, 2022-06-04:
Also happened in 1.6.0b6. Remove USB-C also resolved it.