polyend/TrackerBetaTesting #85: [1.1.16] Sample isn't played from very start in pattern mode or sample playback window

splitradix, 2020-09-23:
I loaded a few samples onto the SD card.

When I import them they play fine in the sample editor window, but in the sample playback and in pattern mode the start of the sample has been moved forward by a few milliseconds, so I'm not hearing the start of the sample. I can

This happens even though I haven't adjusted the start time of any of the samples (start time is 0.000s for everything).

Jacknametrouble, 2020-09-24:
Hey, I cannot reproduce this issue. Can you please zip and link us the particular project where this happens? Thank you.

teamoth, 2020-09-24:
Probably a daft suggestion, but I was wondering if this might be an attack issue? Also I notice that the default attack is 0.010s for instruments and the volume envelope that deals with ADSR is on by default.

splitradix, 2020-09-24:
[bug.zip](https://github.com/polyend/TrackerIssues/files/5278607/bug.zip)
Project attached.

splitradix, 2020-09-24:
Just after I exported the file I noticed that every instrument had an attack setting set to 0.010s despite me not specifying that when loaded them into the new project. This may be human error but I'll try to load a few more samples to see if I can reproduce.

splitradix, 2020-09-24:
> Probably a daft suggestion, but I was wondering if this might be an attack issue? Also I notice that the default attack is 0.010s for instruments and the volume envelope that deals with ADSR is on by default.

I just saw your comment and I am seeing the same thing.

teamoth, 2020-09-24:
If dialling down the attack to zero fixes the issue then I guess the bug to be fixed here in the next version is to have zero attack as the default. Perhaps also the volume envelope shouldn’t be switched on by default?

bambrose, 2020-09-28:
I would agree. I initially didn't clock the 0.010 attack time which renders any percussive samples 'dull'
I would politely suggest probably better to have a default of 0

Stazma, 2020-09-29:
I'm also voting for a default attack time of 0 !

splitradix, 2020-09-29:
> I'm also voting for a default attack time of 0 !

Actually the envelope is technically an effect, like the filter. It should be off by default.

piotr3500, 2020-10-01:
Agreed, Env Attack will be 0 ms by default in 1.1.18.

closed 2020-10-01 by piotr3500