polyend/TrackerBetaTesting #87: Fast/greater increment jogwheel dial - hold shift maybe?

teamoth, 2020-09-24:
I often like to dial in a huge grain length in the granular synth, but it takes a long time to get up to the full length possible, Please can we have a fast jogwheel feature, by pressing shift at the same time maybe?

bambrose, 2020-09-29:
I like this idea.

sebsonf, 2020-10-03:
Same idea l wanted to suggest. :)

bambrose, 2020-10-16:
I love the jog wheel feel on the Tracker, and I have realised it's because it's very similar to the one on the Akai S950 I cut my teeth on.
On that machine one of my favourite parts of the OS was that you scrolled Left/Right to flip between 1000, 100, 10, 1 of any number being edited. That's how it got around moving around big numbers with a single scroll wheel.
So it defaults to increment 1 unit at a time.. press/hold left button to switch to tens *and scroll, left again to select hundreds etc

bambrose, 2021-01-20:
Note that the Scroll wheel is slower on the mk2 vs mk1 machines