polyend/TrackerBetaTesting #88: [1.1.17] In some patterns, Backspace doesn't work in the last two tracks

ambv, 2020-09-24:
### Steps to reproduce

This doesn't happen on every project, not even on every pattern in the affected project. I attached a zipped project that reproduces, also see attached video:

1. First on track 6 I'm demonstrating Backspace working fine.
2. Then on Track 7 and Track 8 I demonstrate that Backspace only moves the cursor up, not changing the contents of the track at all. 
3. However, plot twist, below step 32 (0x20 in the pic) Backspace does work... but it doesn't move steps 32+. 

### Attachments and links

Video: https://www.dropbox.com/s/ro2rwltg19of8qs/BackspaceNoWorkey.mov?dl=0
Project: [myxozoic waltz.zip](https://github.com/polyend/TrackerIssues/files/5277386/myxozoic.waltz.zip)

### Other important information

* How often does this bug happen? every time in some patterns in some projects

* Does this bug happen in previous firmware versions? yes, at least since 1.1.15 when I first noticed

* Does this bug happen in every project? no

Miltos-S, 2020-09-25:
I have come across this bug too, where you can't move or copy-paste steps above 32.
I downloaded this project also to reproduce.
A quick fix I have found for now is to change the pattern length from 96 to 32 and then back to 96 and now backspace works.
For some reason the Tracker 'assumes' that the pattern is only 32 steps and you have to set it again at 96.

ambv, 2020-10-01:
The bug is this:
- when copying tracks in the Song Editor, the length of the track is copied as well
- but the non-copied tracks keep the outdated length
- the Pattern Editor view takes the length from the first track so as long as this one is long enough, it all appears to be fine

closed 2020-10-02 by shuler323