polyend/TrackerBetaTesting #89: [1.16.1] tracker will not read files

cclcng, 2020-09-24:
I have discovered that tracker won’t read any files that are not nested in a sub directory. This is fixed when you make a simple sub directory (like”1) and drag all of your files into it. There is an unstable workaround I found : quickly remove the sd card while in the “empty” directory and plug it back in, and it will look for the files... but this is very temporary and unstable , if you load the files into instrument slots, sometimes they disappear upon reboot.... it seems like it is very reliable to use a sub directory, but since discovering this, I haven’t been able to properly test it. Hopefully y’all can reproduce this show-stopper of a bug. And maybe people who have been struggling with file conversion like I have , will benefit greatly from this !!
Thanks, 
Khonnor

wjakobczyk, 2020-09-26:
Hi @cclcng, unfortunately I can't reproduce the issue. What kind of file are you trying to load from the root directory? Samples? What is your directory structure?

cclcng, 2020-09-26:
I am loading 16 but wav files @44.1 kHz. My folder structure is as follows : 
(Root :/ Samples/ (folder name) 
So, if the files are put directly in this folder without a directory , they fail to load for me , but if it is (root:/samples/ (folder name)/ (1)... *.wav1..*wav2... etc  the same files show up . If you still cannot reproduce this behavior, I will make a small archive of files for you to try with 🙃🙂
Respect ,
Khonnor

wjakobczyk, 2020-09-27:
Yes, please attach an archive with the problematic structure, I still can't reproduce the issue.
And to be sure - you can see the files, but can't load them, right?

cclcng, 2020-09-27:
No, what happens is that the folder does not display the files at all , until you hot swap the sd card. So what I’ll do it make 2 folders of the same collection , one that will read and one that won’t .. ive reproduced this Many manY times 🦴. I will do this  a bit later
Thanks ,
Khonnor

cclcng, 2020-09-27:
OK: Here are two folders, "samples that slap", which should load properly as the samples are in a subdirectory("1") and then there is a folder called "samples that can't slap" which contains the exact same wavs as in the first one,(you might want to check for yourself) the only remarkable difference is that  they havent been moved to a subdirectory. i've gotten many reproductions with  these very files, so... please try again. I might add that my folder strucure is as follows: [I:\(sd card)] [\SAMPLES] [\SAMPLES THAT DO/DON'T SLAP]

it wouldnt let me attach a 7zip archive, so ive uploaded it to my dropbox: [](https://www.dropbox.com/sh/fu0nrhe3gsxr1cs/AADQAhpdpGpRNvGv1lpu0mXBa?dl=0)

I hope this works, or rather, it breaks for you! 
thanks

cclcng, 2020-09-27:
I've now attached an archive with two foolders, one of which reads and one
of which is broken. the only difference netween the two, as you can see, is
the broken one "can't slap" has the files dirrectly in the folder

hope this helps!! i also put this link in the bug repository, but ill give
it here just in case.
https://www.dropbox.com/s/s77tf38mmboupzu/samples.7z?dl=0
P.S. 7Z ARCHIVES USE A SOFTWARE CALLED 7ZIP, IT IS FREE. I HOPE THIS IS NOT
AN ISSUE FOR YOU . ANYWAYS,
BEST,
KHONNOR

wjakobczyk, 2020-09-28:
Thanks, I have tested with your archive.
Unforunately I still can't reproduce the issue - I have put both folders to Samples and can see all content in both, from the Sample loader.

wjakobczyk, 2020-09-28:
Are you sure you are properly ejecting the sdcard from your system after writing the files?
Does this happen to all your folders in Samples, or just this specific collection of samples?

cclcng, 2020-09-28:
I make sure to “safely remove “ hardware every time after I have transferred files. 
The only files it happens with are ones that are not in a subdirectory, but directly in the folder. Wav formatting does not seem to be a factor. 

I want to make sure that the folder sructure is like mine, with a /samples directory within the highest directoryAnd then the folders in question... It is reproducing with me I’ver here without fail.

Jacknametrouble, 2020-09-28:
@cclcng have you tried to remove the hidden "Workspace" folder located in the root directory of the SD card? Or even formatting the SD card? Maybe it's some file structure issues?

cclcng, 2020-09-28:
I had formatted the card (fat32)  In order to start fresh just before I found this issue. I had also deleted the workspace folder (quite a few times) during my testing.

closed 2020-10-12 by piotr3500

ClaidArremer, 2021-04-23:
Just in case anyone is reading this and has a similar problem - I am on 1.3.1 and had the same problem - it's almost certainly to do with the amount of folders/files in one place. I had my entire drum machine collection (50+ folders and thousands of samples) contained in one folder and not everything was showing up. So I deleted the folders I wasn't using and hey presto, now all the ones I kept are showing in the Tracker. ;(