polyend/TrackerBetaTesting #92: [1.1.17] Tracker audio is 16ms behind other synced devices when either master or slave

splitradix, 2020-09-24:
### Steps to reproduce

Sync the Tracker to any other midi device via midi clock and the Tracker's audio will play approximately 16ms behind the other devices. This happens when it is either the master or the slave to midi clock.

### Other important information

* How often does this bug happen? Every time.

* Does this bug happen in previous firmware versions? Yes, in all other firmware versions tested so far.

* Does this bug happen in every project? Yes.

Jacknametrouble, 2020-09-25:
This is a known thing. A latency compensation option will be introduced in the major MIDI update later on the current year.

rpc9943, 2020-09-25:
Good to know it is going to be fixed, when I read the update I thought it fixed latency!

bnckd, 2020-11-23:
same problem here, midi alone runs supertight but as soon as i have samples running as well it's a mess. any news on that major midi update? since summer i can only use the tracker for either midi sequencing or standalone, which is unfortunate, as it was supposed to replace drummachines and sequence modular drums at the same time.

shuler323, 2020-12-22:
try 1.3.0b4, it has implemented first sync delay compensation in PT

closed 2020-12-22 by shuler323