polyend/TrackerBetaTesting #93: Global ADSR adjustment option

splitradix, 2020-09-24:
The ability to change the ADSR envelope of multiple instruments at the same time (applying the same ADSR settings to all of them at once) would be nice.

splitradix, 2020-09-24:
This could also work as an FX maybe, four new effects as attack, decay, sustain, release.

ArtFluids, 2020-09-25:
Yeah, I was really surprised to not see at least decay as an FX location. Even just "fade in" and "fade out" as FX would accomplish sort of the same thing. With linear and exponential curves for fading in/out.

SimmeSimon, 2020-10-11:
Great! The second one could have even more use cases, I think. For example: To have the ADSR thing in the FX-section would also fulfil my desire to shorten and also highlight a sound on just one step, while simultaneously have e.g. a roll effect, so it is more pronounced and clear. 
The only problem I see there: you have multiple (I count six) possible ADSRs (Volume, Cutoff, etc.) So do we want all? which would be 24 (sic) new FX destinations. Maybe even 30 if you also want to account for "Amount" (I could see me using this also) Or maybe just Volume and Cutoff, which would be 8 or 10 possible effects. For me A, D and Amount would be the best options to reduce the list to six and get a good deal of options out of it. (Edit: Wavetable position could yield cool effects here, too. OMG I know that's a lot to ask for. Sorry. I have a lot of gear, but this thing sucked me really knee-deep into a pool of joy and nerdy fiddling with ideas and I have so much fun, which I didn't have in a long time :-D)

I'm aware that I could also use another instrument slot and double the existing one with the altered parameters to achieve this result, but right now I don't know how to copy an instrument with it's existing parameters to another of the 48 slots, so it becomes really tedious to do so.

static-noise-PT, 2021-04-20:
A "time" parameter would be similar and super useful. I mean a parameter which scales the "ADSR" from 1% to 200% or something. I use this kind of stuff very often live and it's a great sound design option.