polyend/TrackerBetaTesting #96: Proposal of the function to Fold tracks.

modmodmodmod, 2020-09-26:
If we hold down shift key in pattern view, we can select multiple tracks we want to Fold in red color with screen key.

After that, press the shift key, Fold will be executed and the screen will return to the normal pattern view.

With this function, for example, only percussive sounds can be combined into one track.  And we can also add 4 pattern fx to the Folded track.

 The contents of Folded track are rendered as a separate audio file.

djkurzman, 2020-10-03:
Just curious, can you elaborate more on 'folding' tracks? It sounds like you are suggesting a function to select multiple tracks and render them to a single audio file/track, which already does exist in record mode in the pattern view.

modmodmodmod, 2020-10-03:
I think the existing audio export feature is a nostalgic ping-pong recording. That's a useful feature, but if It can fold the tracks, we can do the following:

We can finally complete and export the song, and export all the tracks separately when moving to post production such as protools.  (In ping-pong recording, some tracks are combined in monaural.)

 I often would like to use at least one more step Fx per steps.  And if it can fold the track, we can add Fx in an adjustable state.

 In short, I want this feature for non-linear production.

Of course the biggest feature of polyend tracker is that it can realize the classic tracker sequencer in hardware. But I think it's particularly wonderful to come up with a new style by incorporating the random and generative function cultivated in the process of DAW development.

So I'm wondering if the linear effects for the samples and ping-pong recordings could be updated in a more modern way. Personally.

closed 2021-12-02 by Jacknametrouble