polyend/TrackerBetaTesting #98: [1.1.17] Jogwheel acting weird in "Sample Playback" menu

Stazma, 2020-09-27:
Hi there. Since the last firmware revision I noted a weird behaviour of the jogwheel in the Sample Playback menu, especialy when adjusting slice / loop / position points. 
It is like the action of the wheel get "stucked", it has hard time to move even tho the rotation speed applied stay the same. It feel ok on filter frequency or other fader type controls.

https://youtu.be/AS60B39CbbY

I never experienced this in previous firmware versions

This bug happen in every project

Jacknametrouble, 2020-09-27:
Julien, does it happen in Sample Player only, or in all the different sections too?

AirDnB, 2020-09-28:
I experience this in the Sample Edit page.  Turning the jog wheel to adjust Bits/Rate in the Bitcrusher. 

Spinning wheel yields quick movement, followed by slow/no movement. 

Version 1.1.17

https://photos.app.goo.gl/6yFB7pduX1zadSK27

Cheers,

Julian

Stazma, 2020-09-28:
It does happen within the Sample Editor as well (start / end in Crop, delay time). I would say it is everywhere but less noticeable on most controls.

Jacknametrouble, 2020-09-28:
@Stazma @AirDnB guys I have a favour to ask you. Please open up an empty new project and downgrade your firmware version to v1.1.0 for a sec, then check the behaviour of the Jog-wheel in the Sample Editor. Let us know if that changes anything. Thank you,

Stazma, 2020-09-28:
> 
> 
> @Stazma @AirDnB guys I have a favour to ask you. Please open up an empty new project and downgrade your firmware version to v1.1.0 for a sec, then check the behaviour of the Jog-wheel in the Sample Editor. Let us know if that changes anything. Thank you,

I just tried and as well get back to 1.1.17 afterwards and the jogwheel behaviour is mucj better on 1.1.0.

AirDnB, 2020-09-28:
It's way smoother in 1.1.10

The shorter video shows that it does sometimes speed up/pause a little.  

https://photos.app.goo.gl/sT9bmx4BwgiszDpg9

Cheers,

Julian

AirDnB, 2020-09-29:
Hey guys, 

Here's a slightly long video (1:40) showing the interaction with the FM recorder and sample edit pages, that has very pronounced jog wheel 'lag.' 

https://photos.app.goo.gl/GVZJLhoZr7yQ1yYv7

Cheers,

J

Jacknametrouble, 2020-10-01:
@AirDnB @Stazma Guys we're taking a closer look at this issue. Are you sure this only happens in the 1.1.17 on the Sample Playback? The other probability we're considering is a hardware malfunction of the jog-wheel encoders so it's kinda important to us to have a clear message here. Thanks,

Stazma, 2020-10-01:
> 
> 
> @AirDnB @Stazma Guys we're taking a closer look at this issue. Are you sure this only happens in the 1.1.17 on the Sample Playback? The other probability we're considering is a hardware malfunction of the jog-wheel encoders so it's kinda important to us to have a clear message here. Thanks,

After some more checking today it seems that it is the same in any firmware, and getting worse every day, so probably more an hardware issue as you suspect.

piotr3500, 2020-10-01:
Seems like a hardware problem 😤 
Let's discuss next steps via email.