polyend/TrackerBetaTesting #100: Midi input: differente channel can control different instruments

synthexpert, 2020-09-28:
Hi, I owe MPC LIVE and many others standalone machines. The problem with most of them is that MIDI INPUT is limited to a track. Just one. For example if you have a BeatStep Pro and sending CHANNEL 1, 2 and 10 through the MIDI INPUT you will still able to play one  instrument/plugin at a time. All the incoming channels and notes will trigger just one instrument, just one at a time.

Currently the PT is doing something similar: I choose a sample with the sample editor and when receiving midi signal I can just play one instrument at a time. 

MIDI output in the PT is absolutely awesome, **but i would be great to enrich in some extent it's MIDI INPUT capabilities.**

_Even if playing more instruments ad the same time is not possible or not present in the PT original concept **would be still ok to know that INSTRUMENTS 1 is triggered by MIDI CHANNEL 1, INSTR 2 by MIDI CHANNEL 2 and so on._**

PT has outstanding sampling capabilities and WT and GRANULAR functions are amazing, i think it would be great to control its sound with some midi controller/sequencer. Thanks.

sebsonf, 2020-10-08:
This is also something that I noticed yesterday. I have a Nord Drum 3P and was trying to use the tracker as a sequencer, live recording what I was playing on the pads. With the Nord Drum 3P, every pad essentially has its own midi channel (1-6), and its sending the pitch and velocity per channel. I was expecting live recording to handle the incoming signals similar to the polyphony when pressing multiple pads on the tracker. However, I found the following:

- Record of multiple input midi channels not treated as polyphony (distribution on different tracks)
- velocity of the pad strokes was not registered / recorded, micro move however was working fine!

synthexpert, 2020-10-08:
Honestly midi input polyphony is a feature that also many well known machines (MPC...) are missing. And there is a large debate on this among users.

I don't know the grade of complexity to add it to the tracker. But if it will be added will give to this fabulous machine another point of uniqueness. I repeat tracker has rich and well done sound mangling features and would be nice to control them with external sequencers or midi controllers in general. 

My opinion is that the Tracker was originally conceived as a tracker, yes a "simple" tracker. But when we started to use the machine we fall in love with it and its functions and we would like to add anything to it :).

Maybe the guys ad Polyend will develop another machine like a groovebox (yes they are surely able to to this) or they will enrich the tracker to make it the dream machine aided to many kind of users...

VinceKabel, 2020-10-20:
So at the moment you can't route individual midi channels from say the Nord Lead 2x independently to each Polytracker channel? Would be great if you could to make the most of the multitimbrality of the instrument

ROM72, 2020-10-21:
Yes, I wanted to ask for the same enhancement. I would love to assign a specific midi channel per instrument to play pads from instrument 1 on my external midi keyboard and play drums on instrument 2 with external midi drumpads on the same project.

smitheriney, 2021-01-11:
A BILLION TIMES... YES - I WANT A POLYASR10!¬¬ ¬¬¬

smitheriney, 2021-01-11:
multitimbral tracker for the wiiiiiiiinnn

vielfrassss, 2021-02-15:
that would make live drum recording much easier as well.
it would mean that if i used fingerdrumming to record drums, i wouldnt have to record them track by track, i could record all tracks at once.