polyend/TrackerBetaTesting #103: [1.1.17] First time opening large folder of files results in sample loader locking up

bambrose, 2020-09-28:
## Hi, sorry you had trouble!

### Steps to reproduce

Add a large folder of samples to the SD card (large enough that the folder becomes automatically split by the Tracker into folders (1-100), (101-200) etc etc
Go to sample loader. Start previewing samples one by one
First folder is probably OK
go to second folder (101-200)
Repeat process (start previewing samples on by one starting at the top
about halfway down you will see that the preview button stops working..
even exiting that folder and trying with other folders results in silence.

If you're lucky turning the tracker on and off resolves.. i.e. the same folders now work fine.. so linked to the 'conversion' also maybe ?
However one time I did this the samples suddenly sounded very 'gritty' i.e. exceedingly low bit resolution.
IN that instance a reboot didn't resolve, but removing the SD card and re-booting resolved

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
Every time

* Does this bug happen in previous firmware versions?
functionality didn't exist in previous versions

* Does this bug happen in every project?
yes

wjakobczyk, 2020-09-29:
Unfortunately, I can't reproduce it. Does this happen with specific samples only, or any larger folder?

bambrose, 2020-09-29:
Any Large folder I have imported.
you have to preview each of the files in all folders one after the other..
I also have a large number of folders .. and it only happens the first time you open a new folder (then resolves itself after a restart)

wjakobczyk, 2020-09-30:
Ok, thanks, we will try more.

Jacknametrouble, 2020-10-01:
Couldn't replicate anyhow. My test environment > made a folder with over 500 samples (single shot and short loop files with different audio quality files) > got them onto the SD, tracker has split them into subfolders > I went thru every single sample in all the 6 subfolders using Preview + arrows shortcut (also with jog-wheel which get pretty laggy when scrolling fast but only in terms of display, audio works alright) and it worked out perfectly, no freeze, no audio loose.

closed 2020-10-01 by piotr3500