polyend/TrackerBetaTesting #99: [1.1.17] Delete button skips steps when editing during playback

AirDnB, 2020-09-28:
Whilst the pattern was playing I wanted to individually delete about half a track worth of steps (but not all in a row). I started hitting the delete button, but it deletes one step and skips over the next one instead of highlighting/deleting. 

https://photos.app.goo.gl/3qvEcWZ6B8RuK9Vq6

So far only tested in 1.1.17 but happens in two different projects. 

Cheers,

Julian

Jacknametrouble, 2020-09-28:
Was the behaviour different in any of the previous firmware versions?

AirDnB, 2020-09-28:
Works okay in 1.1.0

Jacknametrouble, 2020-09-29:
Hey, a comment to your attached video > you have the Step parameter set to 2 so it's deleting everything every 2 steps, normal behaviour.

closed 2020-09-29 by Jacknametrouble

AirDnB, 2020-09-30:
Yep, that was definitely a mistake from me. Apologies.