polyend/TrackerBetaTesting #106: [v1.1.17] Envelope should be off by default

splitradix, 2020-09-29:
Currently when importing new samples each new instrument has the envelope turned on and the ADSR set to:

Attack: 0.010s
Decay: 0.000s
Sustain: 100
Release: 1.000s

The envelope should be off by default (just like all other effects).

Importing mod files also turns the envelope on with the above settings for each instrument, it should be off.

Miltos-S, 2020-09-30:
The 0.010 Attack does affect some samples losing some punch so it would be better to be at zero by default or even all the envelope turned off.

ArtFluids, 2020-09-30:
I agree, I immediately set all instrument attacks to 0.000s because otherwise all the samples sound weak

piotr3500, 2020-10-01:
Duplicate of #85

closed 2020-10-01 by piotr3500