polyend/TrackerBetaTesting #112: New FX for conditional triggering of a step

ahlstrominfo, 2020-10-01:
Would like a new FX that will play out a step if a conditional is true. Very similar to either conditional trigs on Elektron Digitakt or Step Components on Teenage Engineerings OP-Z.

Similar to the existing Chance FX this will evaluate if a step should be played based upon if an if-statement is true.

Examples of what could be an conditional is, I use capital **I** as the FX name below:

### Play every X times out of Y iterations 
I3:4 Play every third time of four iterations
I2:6 Play every second time of six iterations
This could range from 1:2 to 8:8.

### Play all steps except every X times out of Y iterations
I3!4 Play all steps except every third time out of four iterations.
I1!5 Play all steps except first out of five iterations.
This could range from 1!2 to 8!8.

### Play only if the track to the right or left plays
Ipr Play if the step on the right track would be played.
Ipl Play if the step on the right track would be played.

### Play based upon previous steps
Iips Play if previous conditional has been played
I!ips Play if previous conditional has not been played

### Play only ones
Ipo Play only once during playback. i.e. whenever you stop it will reset and be played next time.

Stazma, 2020-10-01:
This would be great indeed !

OnevenHub, 2020-10-02:
Yes the Elektron trigger conditions are great!

Jacknametrouble, 2020-10-02:
Hey guys, this is a bit more of a Polyend Tracker Github than Elektron gearheads forum :)

ahlstrominfo, 2020-10-02:
> Hey guys, this is a bit more of a Polyend Tracker Github than Elektron gearheads forum :)

That is why I also wrote about the OP-Z that has similar functionality. But I could easily add Deluge to the mix.
All those have great sequencers and it was just a suggestion.

I completely understand if it does not fit in the Tracker-paradigm of sequencers.

Nk-em, 2020-10-03:
> All those have great sequencers and it was just a suggestion.

Now you know how it feels when you suggest something really good but someone else takes only a small part of comment to go off on a tangent.🤣

mroemore, 2020-11-24:
I would also love this feature. It's one of the few sequencer functions from the OP-Z that I wish I had on the Tracker.

A similar OP-Z feature that might also be nice is the conditional triggering of other "step components". In the context of Tracker, this would be turning on/off the ajacent FX column based on the rules state above.

antistatikk, 2021-01-08:
That would be an amazing addition indeed! Thanks for your request.

arphex, 2021-09-14:
this is a great suggestion - and seems to me pretty simple to implement...

closed 2023-07-19 by ahlstrominfo