polyend/TrackerBetaTesting #113: [1.1.17] When I try to load a Sample, Tracker tries to overwrite empty instrument.

modmodmodmod, 2020-10-02:
At this time, multiple files in the SD card are selected with red color.

 The frequency is not every time, but it is frequent.

 So far only one project has been confirmed.

 Please check the video linked.

[https://drive.google.com/file/d/1KG_EZsPuSaxJwdDjRu0Wa9dKlsW9CAUx/view?usp=drivesdk](https://drive.google.com/file/d/1KG_EZsPuSaxJwdDjRu0Wa9dKlsW9CAUx/view?usp=drivesdk)

modmodmodmod, 2020-10-02:
As soon as I continued the production without worrying about it, no sound was played.  This silence can be resolved by rebooting, but it happens many times.

keith-macleod, 2020-10-04:
I've experienced this as well. 1.1.17

piotr3500, 2020-10-05:
Does it happen on every project? Please try creating a new one.

modmodmodmod, 2020-10-05:
Yes. I've already confirmed this in some new projects.  It's not big problem
 to overwrite empty instrument, but I think it's an error.

piotr3500, 2020-10-05:
Roger that. Let's see if it happens in 1.1.18.

closed 2020-10-09 by piotr3500

rahulsavani, 2021-03-27:
I still get the multiple files being highlighted thing in the latest firmware 1.3.0. I assumed it was a feature (since it will load all these files to consecutive instruments), but after searching and playing around I found no way to turn if fully off. I did find that if one presses left and right together it turns the multiple selection into a single selection of the last selected. But when one moves up or down it goes back to a multi selection. Haven't figured out what triggers this different "mode".