polyend/TrackerBetaTesting #114: [1.1.17] Tracks occasionally decide to stop playing

ArtFluids, 2020-10-02:
Once in a while tracks in Tracker seems to stop responding to notes. For example: Track 1 and 4 are working fine, but tracks 2 and 3 just stop responding to any note data in their lanes. It's not like they all stop working, it's only some of them and only some of the time.  I don't know if it's the instruments dropping out or the tracks.
This is with no mutes. No envelopes, no effects. Just plain vanilla pattern mode.

I can't reproduce it, it seems to happen at random. Even if I save a file when the problem is happening everything is fine when I reload. I can't even take a video of it because there's just no way to know when it will happen. But it's happened quite frequently to me.

It seems most likely to happen when I'm editing. And there's nothing I can do to fix it, it just seems like after a while the tracks just decide to work again. Even stopping the pattern and playing doesn't fix it. It just decides to work again after a while.

The two times I've noticed it are: when editing a pattern and when jamming on the pads during a playing pattern. And the weird thing is the tracks/instruments that stop playing are often tracks I'm not even currently messing with.

I tried overloading the pads with notes, seeing if it was some weird issue with the voice count but that doesn't trigger it.

If I notice anything that makes it happen every time I will update you. But for now, know there's some lurking bug that makes tracks decide to stop playing sometimes.

piotr3500, 2020-10-06:
can you reproduce it on 1.1.18?

R6D4, 2020-10-12:
Thanks @ArtFluids 

I am experiencing the exact, same thing.
Some (random?) tracks do have audio dropouts once every while.
Happens in every session at some point I would say.
Like ArtFluids says, it is hard to tell why and when this happens, so reproducing is difficult.
I will report back when I find out more.
I am happy I found someone experiencing the same problem so far at least ;)
I am on 1.1.22

ArtFluids, 2020-10-12:
> can you reproduce it on 1.1.18?

I will try. I haven't used the Tracker for a few days. But it seems like @R6D4 is having a similar issue.

piotr3500, 2020-10-12:
Do you use any midi options when the bug occurs?

R6D4, 2020-10-12:
What midi options are you talking about, external stuff?
Used mine in standalone, no external midi attached.

ArtFluids, 2020-10-12:
No, no MIDI involved

R6D4, 2020-10-12:
OK, just did a Test just playing 7 Tracks, one Pattern, looking at Master page, doing nothing.
The 16th hh disappeared while everything was running for a few minutes, then I grabbed my phone fast
and recorded it. The hh comes in after a few seconds. I am not touching the Tracker.
Sorry for the really low audio, it is late, I am home and my neighbors hate me already ;)
Turn up the volume as high as you can!
Let me know if you need the file, or anything else that might help you.

[IMG_1123.MOV.zip](https://github.com/polyend/TrackerIssues/files/5367506/IMG_1123.MOV.zip)

piotr3500, 2020-10-13:
@R6D4 Can you share the project?

closed 2020-10-14 by andrzejmagierski2