polyend/TrackerBetaTesting #116: LFO speeds in Hz

PEtajaja, 2020-10-02:
I've suggested this quite a while back but thought I would do it again now that everything's being handled through Github.

It'd be great if the LFO's could be non bpm sync'd much the same way that the delay effect can be.

There could maybe be a toggle between LFO syn / not sync and when not sync'd it'd be addressed in Hz.

It'd obviously be better the larger the range and the higher the resolution but if this isn't possible, maybe there's another way the speed in Hz could be addressed such as adjusting the integer / decimal 1 number at a time (000.00)

Hopefully this is clear enough, this would be a pretty huge update for me as it introduces the possibility of creating much more complex rhythms through the LFO.

PaulAntioche, 2020-10-14:
I completely approve of this idea

ROM72, 2020-10-19:
I wanted to open an issue about this too ! Thank you.

jaredarnold, 2020-11-05:
Agree! This was one of my first thoughts on using the Tracker. Crucial for randomness, ambient sounds, etc.

smt923, 2020-11-11:
this would be awesome to have in general and definitely the more range the better

antistatikk, 2021-01-08:
The higher range would be a huge improvement for long, evolving Automation. Great request!