polyend/TrackerBetaTesting #118: [1.17] audio stops working in sample loader

cclcng, 2020-10-02:
### Steps 
I had just started a new project, and began to load samples into instruments, and after previewing 8-10 wavs, the audio cut out, but the tracker wasn’t frozen, as I could reach the different modules etc.
In the pattern module , I could start playback of the pattern but even if I loaded an instrument , it would not play back. I reset config after this, and nothing happened so I rebooted. When I rebooted, I was able to again get preview audio, but once I loaded an instrument I could not play the pads for example , on the sample playback menu. Nothing would happen. So I went to my computer and deleted the workspace folder, and reloaded the card on the tracker , but when I went to the sample loader , I had audio for the first 8 or so wavs, and then it cut out, with the same behavior as before (including working play-pattern) yet no sound from the previously loaded instrument ... 
thinking about reinstalling the firmware and seeing what that does...

### Attachments and links

Often the best way to show the problem is to record a short video,
attach the Tracker project you were working on, and provide links.

Pro-tip: You can drag and drop images and `.zip` files directly to the edit
box when writing your bug reports. Videos are too big as attachments, you
will have to use an external service for this and provide us links. Please
don't use services that delete files after short periods of time.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")

* Does this bug happen in previous firmware versions?

* Does this bug happen in every project?

cclcng, 2020-10-02:
The bug happens every time now. I hadn’t had this  issue since 1.1.15, but it only happened sporadically with that build , and was resolved by 17 (initially, at least).
It happens with every project, but since I had to format my SD last revision, I  lost my old projects, so can’t check with those, only new ones.

cclcng, 2020-10-02:
I’ll add that , upon restart, the sample loader preview works temporarily, but nothing happens when hitting the pads in sample playback mode

cclcng, 2020-10-02:
also, the pads do not work in the save as... module either.

cclcng, 2020-10-02:
Reinstalling the 17 firmware did not resolve the pad issue.

Jacknametrouble, 2020-10-02:
From the 1.1.17 Changelog:
Sample Preview from the pads in the Sample Loader was causing unit freeze and was removed. Same in Sample Editor but only when an Fx is chosen for a preview.

> also, the pads do not work in the save as... module either.

Do you mean that you cannot enter letters in order to name the project?

cclcng, 2020-10-02:
> From the 1.1.17 Changelog:
> Sample Preview from the pads in the Sample Loader was causing unit freeze and was removed. Same in Sample Editor but only when an Fx is chosen for a preview.
>
>
> Do you mean [...]

Correct. And they do not work in the playback module or pattern either.. 

I was able to restore functionality  with an emergency update  to 1.1.17 using the polyend tool.

Jacknametrouble, 2020-10-02:
Thank you for the update. Are you refering to Pattern screen view by the "playback module"?

cclcng, 2020-10-02:
Thank you for the update. Are you refering to Pattern screen view by the "playback module"?

Yes , the pads do not work in the pattern screen, playback screen or even the file>save as... menu after the sound stops within the sample loader screen

pandamaschine, 2020-10-03:
same problem for me

HakkahFR, 2020-10-03:
Same problem, and big big problem

piotr3500, 2020-10-05:
Guys, can you send a video showing when the problems occur?
We have a hard time reproducing the bug.

HakkahFR, 2020-10-05:
The problem didn't happend for me since 2 days now. Very strange

pandamaschine, 2020-10-05:
the problem happened very randomly , it's very difficult to reproduce it :)
in some time when it happened , on sample loader , I preview very fast several samples in a row.

cclcng, 2020-10-05:
I had some luck  with no crashing for a little bit after I did an emergency flash (to 1.1.17) with the polyend tool, and for some time was not able to reproduce it. But now it is back. I will link to the specific samples that it happens with consistently.  To reproduce this error, I would navigate to the folder of waveforms linked below, and hold the preview key whilst  taping the down arrow to scroll. Sometimes I have to use the up arrow after scrolling with tge down arrow to reproduce. Also, any combination of holding the preview button, alternately tapping up/down,  and turning the jog wheel intermittently seems to cause this issue. 

Here are the waveforms that I was previewing: 

https://waveeditonline.com/wav-files.zip

Best,

piotr3500, 2020-10-05:
@cclcng thanks for the package

I managed to replicate it on 1.1.17.
Have zero issues with that on 1.1.18 (out very soon).

closed 2020-10-05 by piotr3500

cclcng, 2020-10-05:
I’m happy you were able to see this !! I cannot reproduce this with other folders of samples it seems 🤬