polyend/TrackerBetaTesting #123: [1.1] Delay send reset to default after turning off and turning on

isbtn01, 2020-10-03:
Hey, 
ive come across this little bug happening only in one of two projects i have ready now. Ive made some patterns then saved the work and turned off the Tracker. When i turned it on again the Delay Send was reset to default even tho in master section everything is visualy set correctly. When i move any paramter in master delay send everything goes back to normal but this happens every time i turn off the machine. When i load the project by opening it through file function the delay send seems to work correctly, the issue is happening only when turning the Tracker off and back on.

Here is the link for the project and a short video with the issue
https://www.dropbox.com/sh/u6xfiap4fi8mdm8/AADgSNaxyYVH7811ezuS5ZGTa?dl=0

How often does this bug happen? (every time, often, "it only happened once")
everytime
Does this bug happen in previous firmware versions?
dont know, my unit came with 1.1
Does this bug happen in every project?
i was able to replicate it in just one of two projets, so im sending you also the whole project

Thank you

Jacknametrouble, 2020-10-03:
Update to 1.1.17

closed 2020-10-03 by Jacknametrouble

isbtn01, 2020-10-03:
yeah i was checking the beta updates out 5 mins ago sorry, will do, thanks