polyend/TrackerBetaTesting #124: [1.1.0] Euclidian Fill Mode

sebsonf, 2020-10-03:
In euclidian fill mode one can fill an equally spaced number of notes k in a repeating pattern of length n .
Example for k = 3 and n = 5

|■|□|■|□|■|■|□|■|□|■|

There could be two options, one with quantization and one without, for the latter micro move fx would automatically be set.

Example for k = 2 and n = 3, set micro move fx to 50 automatically for second note if quantization is off. 

This would also allow for a convenient way to add triplets (k = 3, n = 2) and its relatives. However for k > n, ceil(k/n) tracks would be required for polyphony.

bambrose, 2020-10-16:
excellent idea

wjakobczyk, 2020-12-14:
This functionality has been added in 1.2

closed 2020-12-14 by wjakobczyk