polyend/TrackerBetaTesting #130: [1.1.17] MOD import: High-end frequency content lost

thomasj, 2020-10-04:
When using the Mod Import function, high-end frequency content available in the original files get lost. I attached a hihat sound from a mod, played back in both Polyend Tracker as well as Protracker (using the [Pt2-clone](https://16-bits.org/pt2.php) on macOS, but I think it is pretty accurate to Amiga). In Protracker, this sound is played back at "C3" (in Polyend Tracker it becomes G2 but I guess that is not an error in itself, just a conversion consequence). 
In the original playback, the main sound goes up to about 17kHz until folding appears - whereas in Polyend Tracker it goes to about 8kHz and thereby gets quite muffled.
The same happens in 1.1.0 and for every sound, so it is not mod-specific.

[high-freq-check.zip](https://github.com/polyend/TrackerIssues/files/5322927/high-freq-check.zip)

thomasj, 2020-10-04:
Btw, this would of course be different depending on the Amiga model and whether the lowpass filter is on or off (in the example above, the filter is turned off). But even with an A500 filter turned on the original would have high frequency content up to about 16kHz. 
Maybe this all is outside of the scope of the import. In that case you can close the ticket :)

piotr3500, 2020-10-11:
We'll improve importing MOD files in the future. Thanks for reporting that.