polyend/TrackerBetaTesting #135: Save Instruments

clarte-dot, 2020-10-04:
You spend some time chopping, repitching and adding effects in the sample editor, if you want to benefit from that work in a new project you either have to make written notes on the configuration or delete everything else from the project. It would be fantastic to have the ability to save all those edits and parameters as an instrument for use globally at any time.

nitelfy2020, 2020-11-06:
I agree. That would improve the workflow a lot.

kjnilsson, 2020-12-07:
even being able to import an instrument from another project would be massively useful but a full instrument export / import facility would rock.

bloshko, 2021-01-19:
This will be really useful!

jefffm, 2021-02-17:
Yep, for example re-slicing breaks on every new track is pretty tedious!

I'm getting by with using the "even slices". But some of the Renoise "instruments" I've saved and used/reused for other non-Polyend-Tracker songs are just not feasible to recreate on the PT without being able to save them for reuse somehow.

Jacknametrouble, 2021-12-18:
This has been added in the newest 1.5b1 beta.

closed 2021-12-18 by Jacknametrouble