polyend/TrackerBetaTesting #140: [1.1.17] Polyend bricked - stuck on "please insert sd card to continue"

dimant, 2020-10-04:
Reproduce:

- record sample from radio
- place in pattern
- press play
- remove sd card
- wait maybe 30 minutes (I was cleaning up my samples folder on my desktop)
- reinsert sd card

expected: sd card is detected, everything continues as normal
actual: message to insert sd card remains

steps tried:
- disconnect power, wait a few seconds, reconnect
- other SD card (3 cards: 1x 16gb SDHC, 1x 128gb SDXC, 1x card that came with tracker)
- remove hidden workspace folder
- remove everything
- format SD card with official SD card formatter
- format SD card with guiformatter from FAQ
- use polyend tool to emergency flash 1.1.0 by following instructions in "how to use beta" readme
- use polyend tool to emergency flash 1.1.17

Jacknametrouble, 2020-10-04:
Diman, 

try the following recovery flashing:
- remove the SD card from the Tracker SD card slot,
- power up the Tracker,
- run the Polyend Tool v1.1 (polyend.com/downloads) and detect the Tracker,
- choose the firmware iteration you wan to flash the unit with (rather use v1.1.14 instead of v1.1.0),
- press the hidden reset button located on the back panel of the instrument,
- wait till the progress bar in the app will reach 100%,
- if the Tracker won't reset automatically, reset it manually,
- place the SD card back in the Tracker's SD card port,

Let us know.

Thank you,

J

anemlife, 2020-10-04:
Something similar happened last week

To make the tracker work again here's the method that I used: 

I did a manual emergency update with another SD card (not using polyend tool, I added a "firmware" folder on the new SD containing 1.1.0 firmware, and powered the PT with the new SD while pressing the reset button). 

This worked (polyend tool wasn't working in my case) and I was then able to use my tracker again. I plugged the older SD card back, did an update to 1.1.17 with it and everything was fine.

That said, I don't think you should remove your SD card with the PT on. I always save the project, power PT off and then remove it, and that seems to work!

Let's see if this can help :)

dimant, 2020-10-04:
Thanks! This sounds promising. Could you describe in a little more detail how you do an emergency update from SD card?

I tried:
 - power off
 - insert card with Firmware/PolyendTracker_1.1.0.ptf and nothing else on it
 - keep reset depressed, press power on button

The screen remains black until I release the reset button. When I release it, it goes into the polyend tool "emregency update" mode. Are you doing something differently?

closed 2020-10-05 by dimant

Jacknametrouble, 2020-10-05:
Hey Diman, what was the solution in your case?

dimant, 2020-10-05:
@Jacknametrouble this was a hardware issue, not a bug. Solution is to reach out to Polyend support.

Jacknametrouble, 2020-10-05:
We've received your email today in the morning and replied to it immediately. Please keep us posted.