polyend/TrackerBetaTesting #142: Perform mode: transpose MIDI also

anemlife, 2020-10-04:
Would be cool if the perform mode transpose option could affect MIDI as well.

At the moment it only works with the internal samples. An option to toggle MIDI perform transpose on/off would be perfect so we could keep the PT acting as it is now or with midi transpose on, depending on what we need!

Jacknametrouble, 2020-10-05:
A major MIDI update is planned to drop soon, maybe even in 2020.

anemlife, 2020-10-05:
Yes I heard about it, but didn't knew if the MIDI transpose in perform mode was also planned in the upcoming MIDI update.

Will the update also fix USB sync issue? MIDI sync works perfectly with the MIDI jack input but starts offbeat with USB (you probably already know about it, but just wanted to make sure!)

Thanks for you help, I'm enjoying the PT a lot.

closed 2021-12-02 by Jacknametrouble