polyend/TrackerBetaTesting #145: Proposal of a function that can play the selected range in loop

modmodmodmod, 2020-10-05:
I wish I could edit the steps while looping play back only the selection on the pattern screen.  Please consider it.

modmodmodmod, 2020-10-26:
It would be helpful if I could select multiple slots and play in a loop even in song mode.

TwistedGenetics, 2020-11-25:
I would like this as well, handy on long patterns if you just want to add a few bars of notes and then copy to rest of pattern, or if you want to do a turn around at the end of a pattern and don't want to have wait for it to play though to see if it fits

colonel-blimp, 2022-01-21:
Good ideas, for both pattern and song mode

> It would be helpful if I could select multiple slots and play in a loop even in song mode.

The suggestion for song mode is similar to #598, #763, #1306