polyend/TrackerBetaTesting #147: Multiple instruments selection

anemlife, 2020-10-05:
Maybe that is too complicated or impossible to do, but I would love an option to select/lock let's say 4 instruments, and every line of pads would correspond to 1 instrument's octaves. 

example: 
pads 1-12 => drums 1 from C3 to B3
pads 13-24 => drums 2 from C3 to B3
pads 25-36 => kick 1 from C3 to B3
pads 37-48 => MIDI Channel 1 from C3 to B3

That could be a real game changer for typing a song even faster than you can now. (especially for complex drums, etc...) But I do realize that this I a pretty big feature request, I can totally understand if you don't plan to add such an option! 

Thanks for reading

math0ne, 2020-10-09:
This would be awesome. I think you could achieve something like this now using slices and the slice per pad play mode, but it would be a pain to setup.

anemlife, 2020-10-09:
That would definitely be a pain to setup! That's why an option to simply lock 4 instruments would be so welcome :)

Jacknametrouble, 2021-12-18:
Thank you for your feedback.

closed 2021-12-18 by Jacknametrouble