polyend/TrackerBetaTesting #148: [1.1.17] Filenames starting with # or ! break file navigation

dimant, 2020-10-05:
### Steps to reproduce
- rename a sample to: #sample.wav or !sample.wav
- put it in Samples/
- navigate to samples in sample loader

expected: #sample.wav shows up after . and ..
expected: I am able to navigate to root directory 

actual: #sample.wav appears above . and ..
actual: i cannot navigate to root folder

reproduces 100% of the time

Jacknametrouble, 2020-10-08:
@wjakobczyk do you think it's something we can change?

wjakobczyk, 2020-10-09:
I think it's not trivial, at least. I guess it's a rare case, isn't it?

dimant, 2020-10-17:
@wjakobczyk dunno - I hit the issue when importing a large sample collection onto the polyend SD card. Some of the samples had creative naming strategies trying to appear at the top of the directory listing.

wjakobczyk-polyend, 2021-01-11:
I can't reproduce this in the latest firmware. Might have got fixed with something else.
@dimant have you upgraded your firmware since reporting the issue? Does it also work ok for you now?

liondragon-gh, 2021-08-12:
Similar issue in 1.4.0 b3 with folders names that begin with special characters other than underscore (tested with '-' and '+'): cannot navigate into folders that start w/ special characters.

expected: +[folder] and -[folder] shows up after . and ..
expected: I am able to navigate into the folder

actual: +[folder] and -[folder] appears above . and ..
actual: [enter] behaves like ".." and takes you _up_ one level in the directory structure

trent-boyd, 2022-03-11:
I reproduced this yesterday on firmware version 1.5.0, including the navigation issues. Directories with `!` at the start of the file names appeared before the `../` option. I'm glad to see there's already an open issue for it.

For reference, [here is an ASCII sorting chart](http://support.ecisolutions.com/doc-ddms/help/reportsmenu/ascii_sort_order_chart.htm). All the characters mentioned in this issue appear before `.`. I am assuming that the `../` option is added to the list of directories before sorting happens. The issue might be fixed by adding it to the front of the list _after_ the list has been alphabetized.

closed 2023-04-21 by MiroPoly