polyend/TrackerBetaTesting #149: [1.1.18] In Pattern play and edit mode instruments preview works a few times then doesn't work anymore

synthexpert, 2020-10-05:
## Instrument preview doesn't work in pattern mode (play or edit mode)

### Steps to reproduce

While in pattern mode if I do click the instrument button (the yellow one) I can see on the right the instruments' list.
I can rotate the wheel to select any instrument. I can play the pads to preview the instruments, but unfortunately it works just for 3 or  4 times, after that even if I press the pads I cannot listen any instruments anymore.

Please note: If I click the sample loader the preview function is working fine. The problem is in the pattern mode where you cannot hear the instruments.

(Please fix it it's very important to be able to listen to instruments while editing o playing a pattern.)

### Attachments and links

### Other important information

* How often does this bug happen?  Every time. 
* Does this bug happen in previous firmware versions? I don't know. 

* Does this bug happen in every project? Yes. I restarted the machine many times and tried to create many projects.

stevesomeone, 2020-10-06:
I will confirm this bug. FYI it is the same issue as the following reported on **1.1.17**
https://github.com/polyend/TrackerIssues/issues/118#issue-713760039

In addition - The loss of audition/preview of samples appears after I have been in the Sample Loader/Sample Playback/Sample Edit screens. Once it occurs, there is permanent loss of preview/audition from either the pads or preview button, including the pattern screen.

On several occasions, this has been followed by a total freeze.

It does fix itself after a reboot.
It is random and difficult to reproduce consistently.

synthexpert, 2020-10-06:
Thanks. Honestly it’s not random at all. I reboot the machine as I wrote and the problem is still there, I repeat : in PATTERN edit mode not in Sample load or editor.

closed 2020-10-06 by andrzejmagierski2